sosandroid/AMS_AS5048B #1: Zero position did not reset

MechatronicsWorkman, 2016-04-24:
I was trying to use this code on an Arduino Uno and ran into problems. The read angle code seems to work as expected however the reset zero position did not work for me. After reading the AS5048B data sheet it mentioned to clear the zero position register first before writing a new angle to it. Thus I made the following changes which seem to work.

    void AMS_AS5048B::setZeroReg(void) {
   
    //I ADDED THIS
    AMS_AS5048B::writeReg(AS5048B_ZEROMSB_REG, (uint8_t) 0x00);
    AMS_AS5048B::writeReg(AS5048B_ZEROLSB_REG, (uint8_t) 0x00);
    //I ADDED THIS

    uint16_t zero = AMS_AS5048B::readReg16(AS5048B_ANGLMSB_REG);
    AMS_AS5048B::zeroRegW(zero);
    return;
    }

Just thought I would pass on this information to anyone else using this code.

sosandroid, 2016-04-25:
Hi,

Thanks for this feedback. The issue is solved.

oilXander, 2017-06-03:
Hi,

The issue wasn't solve. The library have some error.

Now there is :

```
void AMS_AS5048B::setZeroReg(void) {
    
	uint16_t newZero = AMS_AS5048B::readReg16(AS5048B_ANGLMSB_REG);
        AMS_AS5048B::zeroRegW((uint16_t) 0x00); //Issue closed by @MechatronicsWorkman
	AMS_AS5048B::zeroRegW(newZero);
	return;
}
```

But for correct work it seems it should look like this

```
void AMS_AS5048B::setZeroReg(void) {

        AMS_AS5048B::zeroRegW((uint16_t) 0x00); //Issue closed by @MechatronicsWorkman
	uint16_t newZero = AMS_AS5048B::readReg16(AS5048B_ANGLMSB_REG);
        AMS_AS5048B::zeroRegW(newZero);
	return;
}
```

sosandroid, 2017-06-21:
Comit done, to be validated as I do not have the device with me right now

closed 2018-02-26 by sosandroid