sosandroid/AMS_AS5048B #6: Zero position seems to not be permanent.

mrkrb, 2017-04-19:
I'm using this library with both a Teensy 3.5 and an Arduino Micro.

Is the `setZeroReg()` method setting permanently the zero position?
In my case the setting is not permanent: if I write the current angle to the `zeroReg` by calling `setZeroReg()` and then I read it by calling `zeroRegR()`, the setting is held by the sensor just in the same session. If I restart the micro-controller and I do not recall `setZeroReg()`, form `zeroRegR()` I read just 0, and not the setting saved before.

Page 28 of the data sheet mention that in order to burn the following procedure is needed:

> 1. Write 0 into OTP zero position register to clear
> 2. Read angle information
> 3. Write previous read angle position into OTP zero position register
> Now the zero position is set.
> If you want to burn it to the OTP register send:
> 4. Set the Programming Enable bit in the OTP control register
> 5. Set the Burn bit to start the automatic programming procedure
> 6. Read angle information (equals to 0)
> 7. Set the Verify bit to load the OTP data again into the internal registers
> 8. Read angle information (equals to 0)

The `setZeroReg()` seems to me to execute only the first three steps, or am I wrong?

Thanks!

brentyi, 2017-05-06:
@mrkrb 
The `setZeroReg()` method only sets the volatile register, but you can set it permanently by burning it to the OTP register with `doProg()`.
This is implemented based as per AMS's recommended procedure for setting the slave address (see #4), but should also work for the zero position.

derpicknicker1, 2017-06-19:
No, the  procedure for setting the slave address (see #4) DOES NOT work for OTP the zero position.

sosandroid, 2017-06-19:
Closed by last PR

closed 2017-06-19 by sosandroid

Moise2901, 2021-07-16:
Hello! Sorry for reopening this thread. I have two questions concerning PR #7 :
1) Say I burn the zero position of my sensor, but later on I need to change it, will this work?
2) If I understand well, I would just need to replace the `setZeroReg()` by the `doProgZero()` command to make zero values permanent?

Thanks in advance for your help!