sosandroid/AMS_AS5048B #9: Arduino Due

baglia, 2018-03-28:
Hi, I tried the library on the UNO and it works perfectly, but when I switch to the DUE I get values(in degrees) only between 270 and 355. I've used the same wiring I used on arduino UNO so I suppose the issue is on the DUE. Does anyone know how to solve the problem ? Thank you in advice!

sosandroid, 2018-04-04:
The Due is known to have some issues with I2C slaves. However, did you use the proper pullup resistors ? What are the results with raw values ?

baglia, 2018-04-04:
hi, thanks for the reply. I've used the pins 20 and 21 that have built in pull up resistors. I'm out now and I have not the raw values but they were limited to an interval as the deg ones.

baglia, 2018-04-04:
I've tried to make an i2c connection between Uno and Due, but they "talk" and after a bunch of bytes they stop instantly. I've ordered  logic level shifters because I think this is a voltage issue. In my project I need to use 5 magnetic encoders to correct the position of my robot arm so I'd like to use a dedicated arduino uno to do all the math and then send it to arduino due witch controls the kinematics. Thank you.

sosandroid, 2018-04-05:
My main purpose was to isolate a communication issue from a math issue.

From what you said, the math issue is not an option.

Just found that issue on the wire lib, this may help understanding the issue : https://github.com/arduino/Arduino/issues/2065

https://github.com/arduino/Arduino/issues?q=is%3Aopen+is%3Aissue+label%3A%22Library%3A+Wire%22

sosandroid, 2018-04-11:
Were you able to push forward your investigation of the issue withe the Due Board ?

baglia, 2018-04-19:
Hi, sorry if I reply late but I was very busy these days. I've tried the logic converter but the arduinos don't communicate at all. I'll keep you updated! thank you for the reply.

closed 2020-02-03 by sosandroid

gseqBE, 2021-07-01:
Please see https://github.com/sosandroid/AMS_AS5048B/issues/18 for a potential fix.