sosandroid/AMS_AS5048B #10: Arduino Uno P

jayjaymomo, 2018-04-10:
Hi there, I was wondering if anyone can tell me what I am doing wrong.

I have a AS5048B connected to my Arduino Uno by doing the following:

 AS5048B                             Arduino uno

 SDA ------------------ >      A4
SCL -------------------->      A5
  
I notice that there are hardware pin such as A1 and A2 located on the AS5048B ... However, I do not know where to put these pins on, furthermore not knowing if I am required to us these pins....

When I go onto my Serial Monitor all I am getting is the following:

Angle degree : I2C error: 2
359.9780273437
 
Any help will be appreciated
Thanks, 
Jess

sosandroid, 2018-04-11:
Hi,

Sharing your code may help.
Sometimes the error 2 means the device is not properly connected. A look at the datasheet may help.

From what I understand, your connection to the device is weak. Did you used some pull-ups resistor on the I2C lines ?

jayjaymomo, 2018-04-11:
Hi there, thank you so much for replying :D

I am currently using your program_address.ino from your example folder to
my Arduino Uno. I read one of your comments in your program_address.ino
saying
 // If A1 & A2 are pulled low, our new address should be 0b1000100

Therefore, I assumed you had to take the A1 and A2 and connected it to GND.
 When trying to use the  i2c protocol for the AS5048B, do we have to
utilize the A1 and A2 pins on the AS5048B?

Furthermore, I have not connected a pull up resistor on my i2c line but I
will do it immediately and see if there are any changes

sosandroid, 2018-04-11:
This example is not the best one to start with the chip. This is part of the advanced features of the chip. You should start with read_simple_angle.ino for instance.

A carefull read of the datasheet will explain you what are the A1 & A2 of the AS50948B board for. This is used to set the chip address.

The pullups are mandatory to get a proper communication over the I2C lines.

jayjaymomo, 2018-04-11:
> Again, I really appreciate your patience.
>
> On Wed, Apr 11, 2018 at 7:23 AM, jessica green <jessica3q42@gmail.com>

4.7kOhm*

jayjaymomo, 2018-04-13:
> 4.7 kOhms should work.
> Does this enhance the communication ?

Haha, I've been banging my head against the table because I can't seem to
get this i2c working even with the 4.7 kohm resistor. My Arduino uno is not
interacting with my the As5048B through the i2c protocol. I kept getting an
I2c error:2 which also means weak connections.

After setting up my 12c communication by using only the SDA and SCL on A4
and A5 pins on the arduino uno while pulling up those pins.. I seem to
always get to the same conclusion with an error of

i2c error:2
(translation: Weak connections)

I will update you if I get the i2c working and let you know if it enhances
the communication
D: sorry haha
- Jess

sosandroid, 2018-04-14:
Can you try something.lower value pull-ups ? 1k for instance ?

jayjaymomo, 2018-04-14:
I try pulling up the SDA and SCL pin with 4.7kohm - received a i2c error:2
so then I tried to pull those pins up with 10kohm  - received a i2c error:2
I don't have 1kohm in stock, so I used a 330 ohm - received a i2c error:2

If I were to leave the A1 and A2 hardware pin alone will that make A1 and
A2 pin Low or 0.
I fully understand that A1 and A2 are the last 2 bits of data of the slave
address so therefore you can either set A1 and A2 as 1 or 0.
?
So my question is.... if you were to leave the A1 and A2 pin alone... is
the slave address still 0x40?

jayjaymomo, 2018-04-14:
Like you suggested, I used the read_simple_angle.ino code to run it on my
Arduino Uno.

sosandroid, 2018-04-15:
Can you sum up your setup and copy/ paste the code and the serial messages provided by the board running ?

jayjaymomo, 2018-04-16:
Hardware I am using:
1. Arduino Uno
2. 330 ohm, 10kohm and 4.7 ohm resistor
3. AS5048B
4. Breadboard
5. electrical jumpers

Rough Schematic:

(330Ohm Resistor)

______________________________

|                                                  |
     Arduino Uno                                 BreadBoard            |
                         AS4058B           |
                 5v-----------------------------------     (+)
--------------------------------------> 5v                   |

    |  <-- (330 ohm) Resistor             |

            |------------------------                       |

                                |                      |

                            V                     |
                  A4----------------------------
--------------------------------------------------------->
SDA                   |
                  A5 ------------------------------
------------------------------------------------------->
SCL                  |

^                                                 |

          |____________________________

setup:
Senerio 1: (Using 330 ohm resistor Pull-up)
Step 1: The As5048B has header pins on them... to i put the AS5048B onto an
breadboard
Step2: I placed a electrical jumper wire connecting Arduino Uno Pin A5 to
SCL pin located on AS5048B (BreadBoard)
Step3: I then placed an electrical jumper wire connecting Arduino Uno pin
A4 to SDA pin located on AS5048B  (BreadBoard)

jayjaymomo, 2018-04-16:
Hardware I am using:
1. Arduino Uno
2. 330 ohm, 10kohm and 4.7 ohm resistor
3. AS5048B
4. Breadboard
5. electrical jumpers

Rough Schematic:

                                    (330Ohm Resistor)

|

V

______________________________

|                                                  |
     Arduino Uno                                 BreadBoard            |
                         AS4058B           |
                 5v-----------------------------------     (+)
--------------------------------------> 5v                   |

    |  <-- (330 ohm) Resistor             |

            |------------------------                       |

                                |                      |

                            V                     |
                  A4----------------------------
--------------------------------------------------------->
SDA                   |
                  A5 ------------------------------
------------------------------------------------------->
SCL                  |

^                                                 |

          |____________________________

GND------------------------------------------------------------------------------------>
GND

// The arduino Code I ran is : reading_simple_angle.ino, when i open the
Serial Monitor at 9600 Baudrate I get:

i2c Error: 2
Angle Degree: 359.99
i2c Error: 2
Angle Degree: 359.99
i2c Error: 2
Angle Degree: 359.99
i2c Error: 2
Angle Degree: 359.99
i2c Error: 2
Angle Degree: 359.99
i2c Error: 2
Angle Degree: 359.99

jayjaymomo, 2018-04-16:
Hardware I used:
BreadBoard
Arduino Uno
1k Ohm resistor
AS5048B
Jumpers
(Magnet that came with the AS5048B on the shaft of the DC Motor)

Comment: Since I soldered Header pins onto my AS5048B, I decided to put the
AS5048B on a BreadBoard

Arduino Uno pin A4 is connected with a jumper to SDA pin on the AS5048B on
the breadboard
Then I connected Arduino Uno pin A5 with a jumper to SCL pin on the AS5048B
on the breadboard

I then connected  the 5v pin on the arduino uno with a jumper to the
breadboard and took two 1kohm resistor and connect one  (1k ohm resistor)
from the 5v line on the breadboard  to the SDA and SCL pin on the AS5048B.
( Therefore, pulling up the SDA and SCL pin with 1kohm resistor)

Thirdly, I then connected the 5v line on the breadboard to the 5v pin on
the AS5048B

Finally, I connected the  GND pin from the arduino uno  to the GND pin on
the AS5048B.

I then uploaded the code simple_angle_read.ino to my Arduino Uno, put the
AS5048B close to my DC motor shaft  and opened up my Serial monitor at 9600
Baudrate
and the Serial monitor outputs the following:

Angle sensor raw: 16383.0000000000
I2C error: 2
Angle degree: 359.9780273437

jayjaymomo, 2018-04-16:
I think i may have bought the wrong part and might have figured out my
problem.... Do you by any chance know if your library will work on an
AS5048A?

jayjaymomo, 2018-04-16:
I solved my answer!!!!! Thank you Thank you so much :dagger:  .... After long hours of starring at my AS5048.... I notice that I accidentally fried my chip .... XD I took my multimeter and notice for some odd reason that no voltage is going through from my arduino to the AS5048B and the reason being was because.... I wasn't paying attention and connected to GND :D

jayjaymomo, 2018-04-16:
In short...I ... kind off short circuited my .... AS5048B -____-''

sosandroid, 2018-04-17:
Bad news,you'll have to find another chip :)

closed 2018-04-17 by sosandroid