sosandroid/AMS_AS5048B #12: Random input from sensor

jodege, 2019-06-03:
Problem:
Sensor angle degree readout constantly fluctuates between roughly +/-13 degrees ....even though complete idle/resting !?
And Impossible to gain any degree values over 90 ?? (Rollover to Zero at about 90 degree)

Any help on this would be tremendously appreciated, as stuck in development for the moment !

Sidenotes:
No closeby stray magnetic fields exists & no metal parts near sensor.
Radially polarized magnet used (came with the AMS board)

Setup:
* DUE (pins: SDA/20, SCL/21).
* Functionally Non-modified/original "read_simple_angle" Sketch (enclosed).
* 5V & 3.3V on AS5048B shorted (Feed 5V from DUE).
* 2.2uF cap over VCC/Gnd.
* Both SDA+SCL pulled up by each 15k (even though already done by DUE. Also tested without but no difference)

**** (original) Sketch used
#include <ams_as5048b.h>
#define U_RAW 1
#define U_TRN 2
#define U_DEG 3
#define U_RAD 4
#define U_GRAD 5
#define U_MOA 6
#define U_SOA 7
#define U_MILNATO 8
#define U_MILSE 9
#define U_MILRU 10

AMS_AS5048B mysensor;

void setup() {
  Serial.begin(9600);
  while (!Serial) ;
  mysensor.begin();
  mysensor.setZeroReg();
}

void loop() {
  Serial.print("Angle sensor raw : ");
  Serial.println(mysensor.angleR(U_RAW, true), DEC);
  Serial.print("Angle degree : ");
  Serial.println(mysensor.angleR(U_DEG, false), DEC);
  delay(2000);
}

Sample monitor readout_1:
raw : 2250.0000000000   degree : 49.4384765625
raw : 2570.0000000000   degree : 56.4697265625
raw : 2314.0000000000   degree : 50.8447265625
raw : 2378.0000000000   degree : 52.2509765625
raw : 2634.0000000000   degree : 57.8759765625
raw : 2314.0000000000   degree : 50.8447265625
raw : 2122.0000000000   degree : 46.6259765625
raw : 2442.0000000000   degree : 53.6572265625
raw : 2122.0000000000   degree : 46.6259765625
raw : 2314.0000000000   degree : 50.8447265625
raw : 2378.0000000000   degree : 52.2509765625
raw : 2122.0000000000   degree : 46.6259765625
raw : 2250.0000000000   degree : 49.4384765625
raw : 1994.0000000000   degree : 43.8134765625
raw : 2442.0000000000   degree : 53.6572265625

Sample monitor readout_2:
raw : 158.0000000000   degree : 3.4716796875
raw : 478.0000000000   degree : 10.5029296875
raw : 414.0000000000   degree : 9.0966796875
raw : 158.0000000000   degree : 3.4716796875
raw : 4062.0000000000   degree : 89.2529296875
raw : 414.0000000000   degree : 9.0966796875
raw : 606.0000000000   degree : 13.3154296875
raw : 350.0000000000   degree : 7.6904296875
raw : 286.0000000000   degree : 6.2841796875
raw : 542.0000000000   degree : 11.9091796875
raw : 350.0000000000   degree : 7.6904296875
raw : 30.0000000000   degree : 0.6591796875
raw : 414.0000000000   degree : 9.0966796875
raw : 606.0000000000   degree : 13.3154296875
raw : 286.0000000000   degree : 6.2841796875
raw : 286.0000000000   degree : 6.2841796875
raw : 414.0000000000   degree : 9.0966796875
raw : 4062.0000000000   degree : 89.2529296875
raw : 606.0000000000   degree : 13.3154296875
raw : 286.0000000000   degree : 6.2841796875

jodege, 2019-06-04:
Tested also on UNO = Working like a clock !
But still would be a very useful if possible to get the DUE running (still no clue)

sosandroid, 2019-06-04:
Hi,

Sorry to hear this does not work on Due board. It seems to be equivalent to issue#9.

As far as I remember, my search through the forums at that time, the low level TWi library for due was suspected but I had no way to make it clear.

Hope this helps but I have not anymore a board nor sensor available for testing

jodege, 2019-06-07:
Thanks for your time commenting!
I will comment in case I manage solve but it seems a tough challenge.

closed 2020-02-03 by sosandroid

gseqBE, 2021-07-01:
Please see https://github.com/sosandroid/AMS_AS5048B/issues/18 for a potential fix.