sosandroid/AMS_AS5048B #13: Support esp32 possible ?

froussel, 2019-09-01:
Hello,
It would be nice if this library works with an ESP32
Is it possible ?

sosandroid, 2019-09-01:
I have no more spare parts to test not ESP32
In theory, this works. Did you test ?

froussel, 2019-09-01:
Yes if you can hepl me :-)
Je continue en français...
J'ai plusieurs esp32, j'ai vu qu' on pouvait attribuer sur les esp 32 et 8266 les pins que l'on veut en mettant 
wire.begin ( SDA, SCL, voire vitesse );  ......pas testé
mais j'ai testé avec un MPU 9250 en mettant juste wire.begin 
                                 SDA   SCL
 pour TTGO                23     22           
 pour MH-et-live        21     22            
 pour lolin32 lite         15      2     attention pour téléverser déconnecter 2
 pour lolin32 lite          0      4     aussi
   esp32 doit 30 pins  21     22
 pour lolin32 OLED         0      4
 pour lolin32 oled         26     25    aussi

#define SDA_PIN 0             // lolin lite inscrit 15 c'est 21
#define SCL_PIN 4               //  lolin lite inscrit 2 c'est 22

C'était y a quelques mois...
J'ai aussi un colistier qui m'a aidé pour utiliser le capteur sur un nano mais en SPI, cela  fonctionne bien 👍 
 Je voudrais passer à l'esp 8266 pour cette application j'ai en stock du AMS5048 A ET B mais pour plus tard l'esp32 est mieux.
Donc si je ne suis pas newbee sur arduino IDE, mon niveau ne me permet que de modifier des exemples qui fonctionnent déjà sur ma table ;-) et pas d'entrer dans une bibliothèque

Google translate 
I continue in French ...
I have several esp32, I saw that one could attribute on the esp 32 and 8266 the pines that one wants while putting
wire.begin (SDA, SCL, or even speed); not tested
but I tested with an MPU 9250 by just putting wire.begin
                              SDA SCL
 for TTGO                 23 22
 for MH-and-live      21 22
 for lolin32 lite          15 2 caution to upload disconnect 2
 for lolin32 lite           0 4 too
  esp32 needs 30 pins 21 22
 for lolin32 OLED       0 4
 for lolin32 oled         26 25 also

#define SDA_PIN 0      // lolin lite registered 15 it's 21
#define SCL_PIN 4      // lolin lite registered 2 it's 22

It was a few months ago ...
I also have a mate who helped me to use the sensor on a nano but in SPI, it works well 👍
 I would like to upgrade to esp 8266 for this application I have in stock of AMS5048 A ET B but for later the esp32 is better.
So if I'm not newbee on arduino IDE, my level only allows me to modify examples that already work on my table ;-)

froussel, 2019-09-01:
Pour l'instant l'exemple read_simple_angle.ino   compile.... Me faudrait un schéma de branchement entre l'esp32 et l'ams5048B  je n'ai que la puce et l'aimant.

For now the read_simple_angle.ino example compiles .... I need a schematic connection between the esp32 and the ams5048B I only have the chip and the magnet

sosandroid, 2019-09-01:
Pour le branchement, c'est simple... SDA sur SDA, scl sur scl avec le plus et la.masse.
Attention à être sur un voltage compatible

sosandroid, 2019-09-01:
Ne pas oublier les résistances pullup entre SDA et le.plsu puis scl et le plus. Prendre 1k

froussel, 2019-09-01:
Sur mon montage  précédent avec un arduino : pullup 470 ohms avec le 3v.
Celui qui m'avait aidé m'avait parlé de la longueur des 4 fils qui influençait... Je n'avais pas bien compris mais il me semble me souvenir que c'est la valeur de ces résistances qu'il fallait adapter.
Et surtout dans l'appel il serait intéressant pour pouvoir utiliser plusieurs AS5048B de sortir la variable AS5048_ADDRESS qui peut permettre d'avoir jusqu'à 4 puces à la fois.

Je teste avec un ESP8266

sosandroid, 2019-09-01:
Allez y avec la résistance de 470. 
Regardez le constructeur de la librairie (début du .cpp), il est possible de spécifier l'adresse lors de la création de l'objet

froussel, 2019-09-01:
Test avec un esp8266 wemos D1 qui doit avoir SCL = GPIO  soit D1 et SDA = GPIO4 soit D2
exemple read_simple_angle.ino
A1 et A2 à gnd sur l'AS5048B
1) compile parfaitement
2 ) message en port série en boucle :
Angle sensor raw : I2C error: 2
16383.0000000000
Angle degree : 359.9780273438

sosandroid, 2019-09-02:
C'est un code erreur renvoyé par la librairie Wire. Problème de communication

froussel, 2019-09-02:
Merci,
J'ai dû cramé la puce avec mes soudures... C'est petit !
Je reviens quand j'ai refait le montage

froussel, 2019-09-02:
J'ai dû cramé la puce; 
Je reviens ici quand j'ai refait le montage

froussel, 2019-09-05:
Test positif sur une puce neuve en modifiant la ligne 60 dans ams_as5048b.cpp 
au lieu de wire.begin mettre wire.begin( SDA, SCL);  où SDA et SCL sont les n° des GPIO choisis 
D1: 5, D2:4, D3 : 0 ou D4: 2 sur esp8266 wemos D1, j'ai pas essayé les autres, mes lectures m'ont laissé en mémoire que toutes les pins fonctionnent sauf une.

Si vous étiez sympa : 
- sortir le wire.begin de la librairie car on n'aurait pas à modifier la librairie juste le faire  dans mysensor.begin(SDA,SCL); par exemple
- idem mettre dans l'appel de la lecture les valeurs de l'adresse de la puce à qui on s'adresse car suivant la manières de connecter A1 et A2 on a 4 valeurs possibles donc la possibilité d'avoir 4 encodeurs utilisables.

Dans https://github.com/crazytiti/Treuil-Handi mon pro m'avait codé la possibilité de compter les tours car la poulie devait faire plusieurs tours et l'AS5048B avait l'aimant sur l'axe de la poulie.
On a sur l'esp8266 et l'esp32 aussi la possibilité d'émuler une EEProm mais y a aussi d'autres manières de stocker une variable qui sera conservée même en cas de redémarrage soit par coupure alim soit par déclenchement de watchdog. En l'intégrant dans la librairie cela pourrait aider du monde.

Je teste dès que j'ai un moment sur esp32 mais à première vue pas de raison que cela ne fonctionne pas.
Et encore merci pour le travail.

sosandroid, 2019-09-06:
Bonjour,

Pour le support de l'esp8266, il suffit de commenter la ligne 61 du fichier de header de la librairie. Cela permet de sortir le wire.begin()

Pour le support de plusieurs capteurs, regardez l'exemple xplane_dials, il démontre une des manières de faire.

Pour la mémoire, regardez les autres libs de mon dépôt, il y a la lib Fram_mb85rc
https://github.com/sosandroid/FRAM_MB85RC_I2C, C'est aussi sur le bus I2C

froussel, 2019-09-08:
Ok, c'est parfait 
Merci

closed 2019-09-08 by froussel