sosandroid/AMS_AS5048B #14: simple question i2c error 2

goephs, 2020-10-02:
Using the device with arduino uno set up with i2c connections, I only see the error message: Angle sensor raw : I2C error: 2
Is there another setting that should be used?

sosandroid, 2020-10-03:
Hello,

Have a look into the code, the angleR() and readReg16() may help.

The error is sent by the last one when an I2C communication error occurs. Maybe you need to check your setup

goephs, 2020-10-03:
Many thanks for your tips.  I used an i2c scanner routine and saw that the UNO showed 0x44 (once) and then 0x45.

I changed the .h file to comment out two lines and make address changes:

 

// Default addresses for AS5048B

 

//#define AS5048_ADDRESS 0x40 // 0b10000 + ( A1 & A2 to GND)

#define AS5048_ADDRESS 0x45    // new address

 

#define AS5048B_PROG_REG 0x03

 

//#define AS5048B_ADDR_REG 0x15

#define AS5048B_ADDR_REG 0x44 // new address

 

The device now works without errors and the read raw angle program is reporting results to me.

 

Regards,

Gavin Watson

sosandroid, 2020-10-04:
The best way to change the device address into the code is not modifying the class but modifying the object instance call :

AMS_AS5048B mysensor(0x45);

However, this works ... But if you have more than one sensor to deal with this cannot work.

closed 2020-10-04 by sosandroid