sosandroid/AMS_AS5048B #21: undefined reference to `AMS_AS5048B::begin()'

BirBDey, 2024-11-01:
I am trying to compile the code, but it throws an error saying:
.pio\build\due\src\main.cpp.o: In function `_GLOBAL__sub_I_pos':
main.cpp:(.text.startup._GLOBAL__sub_I_pos+0x4): undefined reference to `AMS_AS5048B::AMS_AS5048B()'
.pio\build\due\src\main.cpp.o: In function `setup':
main.cpp:(.text.setup+0xe): undefined reference to `AMS_AS5048B::begin()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\due\firmware.elf] Error 1

BirBDey, 2024-11-05:
The CPP file needs to be in "src" and the header file needs to be in "include" folder.

closed 2024-11-05 by BirBDey

sosandroid, 2024-11-06:
Thanks for your feedback 
At the time of writing this library, there was no platformIO plan :)